frangoteam/FUXA #309: Bugs in imported views

jcxjcx123, 2022-03-23:
Hi!
I found a problem, I exported a view yesterday and I wanted to import it into the project today, but when I imported it, I found that the view did not display correctly and the view window kept flashing, is this a BUG?
[image]

unocelli, 2022-03-27:
Hi,
I can't reproduce it...It's the issue in the view that you have exported? can you share me?

jcxjcx123, 2022-03-28:
hi,
ok,sure,thanks for your help.
[josn.txt](https://github.com/frangoteam/FUXA/files/8359329/josn.txt)

jcxjcx123, 2022-03-28:
ok,sure.
thanks for your help.
[josn.txt](https://github.com/frangoteam/FUXA/files/8359334/josn.txt)

unocelli, 2022-03-30:
I think that the problem is that you will import a view but you give a project file. 
Export a view:
[image]

unocelli, 2022-03-31:
Yes, the name is fuxa-view.json and not fuxa-project.json how you show in the first comment

jcxjcx123, 2022-04-01:
Oh! Sorry,I's my mistake,I imported a project file instead of a view file. I also have a question. I noticed that the SVG shapes code in Fuxa was very short and rendered very well, but the graphics I exported with Inkscape tended to be very complex and very long. So I would like to ask you what tools you usually use to draw SVG shapes?

unocelli, 2022-04-01:
It depends, I also use inkscape, which is full featured, for simpler things I use [diagrams.net](https://www.diagrams.net/).

jcxjcx123, 2022-04-02:
Thank you very much for your answer Sir

closed 2022-04-02 by jcxjcx123