frangoteam/FUXA #310: Error: Cannot find module 'node-sass'

studuino, 2022-03-24:
Hi, 
I cannot compile fuxa 1.1.5.
I am having issues as below.
Is there anything I am missing?

Environment:
Angular CLI: 7.3.10
npm : 6.14.15
Node: 14.18.0
OS: win32 x64(Win10-64bit)

ERROR in ./theme.scss
Module build failed (from ../node_modules/mini-css-extract-plugin/dist/loader.js):
ModuleBuildError: Module build failed (from ../node_modules/sass-loader/lib/loader.js):
Error: Cannot find module 'node-sass'

<details>
  <summary>Click For Error Details</summary>
  PS D:\FUXA-1.1.5\client> npm start

> fuxa@1.1.5-625 start D:\FUXA-1.1.5\client
> ng serve

** Angular Live Development Server is listening on localhost:4200, open your browser on http://localhost:4200/ **
                                                                                          
Date: 2022-03-24T12:26:34.204Z
Hash: 85ea37a285292ba2072d
Time: 54363ms
chunk {main} main.js, main.js.map (main) 1.93 MB [initial] [rendered]
chunk {polyfills} polyfills.js, polyfills.js.map (polyfills) 309 kB [initial] [rendered]
chunk {runtime} runtime.js, runtime.js.map (runtime) 6.08 kB [entry] [rendered]
chunk {scripts} scripts.js, scripts.js.map (scripts) 42.2 kB [entry] [rendered]
chunk {styles} styles.css, styles.css.map (styles) 93.2 kB [initial] [rendered]
chunk {vendor} vendor.js, vendor.js.map (vendor) 11.2 MB [initial] [rendered]

ERROR in ./theme.scss
Module build failed (from ../node_modules/mini-css-extract-plugin/dist/loader.js):
ModuleBuildError: Module build failed (from ../node_modules/sass-loader/lib/loader.js):
Error: Cannot find module 'node-sass'
Require stack:
- D:\FUXA-1.1.5\client\node_modules\sass-loader\lib\loader.js
- D:\FUXA-1.1.5\client\node_modules\loader-runner\lib\loadLoader.js
- D:\FUXA-1.1.5\client\node_modules\loader-runner\lib\LoaderRunner.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\NormalModule.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\NormalModuleFactory.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\Compiler.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\webpack.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-webpack\src\webpack\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-webpack\src\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-angular\src\dev-server\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\architect\src\architect-legacy.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\architect\src\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\models\architect-command.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\commands\build-impl.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\schematics\tools\export-ref.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\schematics\tools\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\utilities\json-schema.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\models\command-runner.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\lib\cli\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\lib\init.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\bin\ng
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:93:18)
    at Object.sassLoader (D:\FUXA-1.1.5\client\node_modules\sass-loader\lib\loader.js:46:72)
    at D:\FUXA-1.1.5\client\node_modules\webpack\lib\NormalModule.js:301:20
    at D:\FUXA-1.1.5\client\node_modules\loader-runner\lib\LoaderRunner.js:367:11
    at D:\FUXA-1.1.5\client\node_modules\loader-runner\lib\LoaderRunner.js:233:18
    at runSyncOrAsync (D:\FUXA-1.1.5\client\node_modules\loader-runner\lib\LoaderRunner.js:143:3)
[... 199 lines cut ...]
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\NormalModule.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\NormalModuleFactory.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\Compiler.js
- D:\FUXA-1.1.5\client\node_modules\webpack\lib\webpack.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-webpack\src\webpack\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-webpack\src\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-angular\src\dev-server\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\architect\src\architect-legacy.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\architect\src\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\models\architect-command.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\commands\build-impl.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\schematics\tools\export-ref.js
- D:\FUXA-1.1.5\client\node_modules\@angular-devkit\schematics\tools\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\utilities\json-schema.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\models\command-runner.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\lib\cli\index.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\lib\init.js
- D:\FUXA-1.1.5\client\node_modules\@angular\cli\bin\ng
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:902:15)
    at Function.Module._load (internal/modules/cjs/loader.js:746:27)
    at Module.require (internal/modules/cjs/loader.js:974:19)
    at require (internal/modules/cjs/helpers.js:93:18)
    at Object.sassLoader (D:\FUXA-1.1.5\client\node_modules\sass-loader\lib\loader.js:46:72)
i ｢wdm｣: Failed to compile.
</details>

Cigeon, 2022-03-24:
Hi, I am also using node 14 and it's works for me 
`npm uninstall node-sass
npm install node-sass@4.14.1`

studuino, 2022-03-26:
Hi @Cigeon 
Thanks.
I am not so sure but the problem comes from different python versions.
I use pyenv-win to switch between different versions.
Now the problem should have been solved.

<details>
<summary>Click for error details</summary>
PS D:\FUXA-1.1.5\client> npm install

> node-sass@4.13.1 install D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-angular\node_modules\node-sass
> node scripts/install.js

Downloading binary from https://github.com/sass/node-sass/releases/download/v4.13.1/win32-x64-83_binding.node
Cannot download "https://github.com/sass/node-sass/releases/download/v4.13.1/win32-x64-83_binding.node": 

HTTP error 404 Not Found

Hint: If github.com is not accessible in your location
      try setting a proxy via HTTP_PROXY, e.g.

      export HTTP_PROXY=http://example.com:1234

or configure npm proxy via

      npm config set proxy http://example.com:8080

> node-sass@4.13.1 postinstall D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-angular\node_modules\node-sass
> node scripts/build.js

Building: D:\nvm\v14.18.0\node.exe D:\FUXA-1.1.5\client\node_modules\node-gyp\bin\node-gyp.js rebuild --verbose --libsass_ext= --libsass_cflags= --libsass_ldflags= --libsass_library=
gyp info it worked if it ends with ok
gyp verb cli [
gyp verb cli   'D:\\nvm\\v14.18.0\\node.exe',
gyp verb cli   'D:\\FUXA-1.1.5\\client\\node_modules\\node-gyp\\bin\\node-gyp.js',
gyp verb cli   'rebuild',
gyp verb cli   '--verbose',
gyp verb cli   '--libsass_ext=',
gyp verb cli   '--libsass_cflags=',
gyp verb cli   '--libsass_ldflags=',
gyp verb cli   '--libsass_library='
gyp verb cli ]
gyp info using node-gyp@3.8.0
gyp info using node@14.18.0 | win32 | x64
gyp verb command rebuild []
gyp verb command clean []
gyp verb clean removing "build" directory
gyp verb command configure []
gyp verb check python checking for Python executable "python2" in the PATH
gyp verb `which` failed Error: not found: python2
gyp verb `which` failed     at getNotFoundError (D:\FUXA-1.1.5\client\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (D:\FUXA-1.1.5\client\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (D:\FUXA-1.1.5\client\node_modules\which\which.js:80:29)
gyp verb `which` failed     at D:\FUXA-1.1.5\client\node_modules\which\which.js:89:16
gyp verb `which` failed     at D:\FUXA-1.1.5\client\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at D:\FUXA-1.1.5\client\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqCallback.oncomplete (fs.js:191:21)
gyp verb `which` failed  python2 Error: not found: python2
gyp verb `which` failed     at getNotFoundError (D:\FUXA-1.1.5\client\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (D:\FUXA-1.1.5\client\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (D:\FUXA-1.1.5\client\node_modules\which\which.js:80:29)
gyp verb `which` failed     at D:\FUXA-1.1.5\client\node_modules\which\which.js:89:16
gyp verb `which` failed     at D:\FUXA-1.1.5\client\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at D:\FUXA-1.1.5\client\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqCallback.oncomplete (fs.js:191:21) {
gyp verb `which` failed   code: 'ENOENT'
gyp verb `which` failed }
gyp verb check python checking for Python executable "python" in the PATH
gyp verb `which` succeeded python C:\Users\win10\.pyenv\pyenv-win\shims\python.BAT
gyp ERR! configure error 
gyp ERR! stack Error: Command failed: C:\Users\win10\.pyenv\pyenv-win\shims\python.BAT -c import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack   File "<string>", line 1
gyp ERR! stack     import sys; print "s. sys.version_info[:3];
gyp ERR! stack                                               ^
gyp ERR! stack SyntaxError: EOL while scanning string literal
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:383:12)
gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
gyp ERR! stack     at maybeClose (internal/child_process.js:1058:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:293:5)
gyp ERR! System Windows_NT 10.0.19044
gyp ERR! command "D:\\nvm\\v14.18.0\\node.exe" "D:\\FUXA-1.1.5\\client\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--verbose" "--libsass_ext=" "--libsass_cflags=" "--libsass_ldflags=" "--libsass_library="
gyp ERR! cwd D:\FUXA-1.1.5\client\node_modules\@angular-devkit\build-angular\node_modules\node-sass
gyp ERR! node -v v14.18.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
</details>

closed 2022-03-26 by studuino