frangoteam/FUXA #314: Actions for internal tags

AlexandruAndrei3, 2022-03-30:
Hi,

I have tried your tutorial for scripts. It works great for external tags but internal tags don't seem to work. Did it test it for internal tags as well?

unocelli, 2022-03-30:
Hi,
If you mean internal device, it's only frontend, the script run in backend. you can use the server tags.
I planed to implement the cyclically functionality.

AlexandruAndrei3, 2022-03-31:
I tried to assign a value to an internal tag and got no results. Do I understand that operations with internal tags cannot be performed in a script function?
[image]

I consider that the facility to perform operations with internal tags as well as the transfer of values between external tags to internal ones and vice versa is very important.
Do you think that it is possible to perform these operations in the future?

unocelli, 2022-03-31:
the idea would be to have variables for logic only on the frontend, for this internal (I have to change the name). Whereas in the latest version it is possible to define tags on the device server, which are also accessible on the backend.
[image]

robsori, 2022-04-03:
Yes, I noticed in the latest version that we can create tags on the server device. I called them in a simple function and tried to gather their values. The problem is, they can't get any results. Am I wrong somewhere?
I think it is important to be able to make the connection between the server tags and the internal tags.

unocelli, 2022-04-03:
have you tried creating a new project to test the server tags? it is possible that projects made with older versions do not enable the server to check tags. I have to investigate.

robsori, 2022-04-03:
Thank you for your answer. Okay, I'll investigate tomorrow and tell you what I got.

robsori, 2022-04-04:
I tested the latest version of Fuxa and I confirm that the mathematical operations performed between two tags declared on the server side and called in a function is possible. Thanks.

unocelli, 2022-04-05:
Great,
so we can forget about giving the internal tags to the server. :wink:

Rainproof, 2022-05-08:
Yes. I also first tried to work with internal variables. Then I read these messages and made tags on the server. Everything is working. All mathematical functions are executed.

unocelli, 2022-07-24:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-24 by unocelli