frangoteam/FUXA #318: Mathematical operations

AlexandruAndrei3, 2022-04-04:
Hello,

FUXA currently has the option to do basic mathematical operations to tags? Like in the image below? 
[image]
I think it would be useful to do basic mathematic operations inside proprieties of the output control and bar graph for better overall control of tags.

unocelli, 2022-04-05:
Hi,
I suppose the only operation to do is division to convert a integer to a decimal. so I think it would be simpler to add the division operation like the modbus into the tags property.
[image]

AlexandruAndrei3, 2022-04-06:
Thanks for your answer, but I was wondering if there is something similar for SiemensS7? Or if it will be added?
[image]

unocelli, 2022-04-07:
I should add it

unocelli, 2024-10-27:
This feature is available with Read/Wrirte Scale Script

closed 2024-10-27 by unocelli