frangoteam/FUXA #319: Siemens PLC - how to get a bit from a word?

IonuDragomir1, 2022-04-05:
Hi,
From a Siemens S7-1500 PLC I was trying to access a bit from a word. In Fuxa I declared a bool variable. As you can see in the image below, instead of using a true/false value for bit 0.2, I get the 1024 value (2^10). 
[image]
What should be done to get the bool value in Fuxa?
This is how I get in Fuxa the value of the entire word and the value of bit 0.2:
[image]

unocelli, 2022-04-05:
Hi,
currently only one data type can be addressed per address (word or boolean), for which functionality you need to address different type?

IonuDragomir1, 2022-04-05:
Hi,
For example for an Auma actuator I have a status word (16 bits). It contains informations about the actuator. Often is necessary to get, for example, bit number 3 from status word. Bit number 3 could mean that the actuator is running. This is why I need the value of bit 3 ( true or false) and not the actual value of status word. Is there a way to see the value of all bits of status word in fuxa? I mean if I could extract bits from the status word.

unocelli, 2022-04-05:
Yes, but in that case you can address the word with booleans for ex. db2.dbx0.2, why do you need word db2.dbw0 as value?
to filter bits of word (Bitmask) is in progress #295

IonuDragomir1, 2022-04-05:
The word value was displayed for test purpose.  I agree that the bit could be addressed as db2.dbx0.2, but my question is why I get value 1024 into a boolean tag?

unocelli, 2022-04-05:
I group the addresses to minimise the traffic, the same address for different type creates conflicts that are not managed. 
I admit there is potential for improvement.

IonuDragomir1, 2022-04-05:
I'll be following with big interest your updates, especially the ones discussed in #295 . In automation splitting a word/dword in bits is often used. Many thanks!

unocelli, 2022-10-12:
This works in the latest version. let me know if you have any issues.

closed 2022-10-12 by unocelli