frangoteam/FUXA #320: Changing Views - any way to change view accordinly to Tag changes

AgrisNN, 2022-04-07:
Dear Fuxa Team,

My project has several views, one of them is set as "default" and is displayed on /home page.
The project also has few tags, linked to OPC server. Tags are updated correctly.
Once a tag changes a value, I need to change View.
But there's now way to do it - nor with Scripts, Alarms or anything else.
The only action to change View is available inside Button element.

Currently I have to use GreaseMonkey to monitor input field with tag linked and "re-click" a button.
Not a "clean" solution...

Help appreciated.

unocelli, 2022-04-07:
Hi,
I was curious about the feature...so I added...in what context do you use it?

AgrisNN, 2022-04-08:
Hi unocelli,

To clarify and satisfy curiosity: there;s a machine with hardware buttons being read by PLC and appropriate tag is being read by Fuxa over OPCUA. 
Once human presses a button, Fuxa must change View.
Two pics below shows how it looks like now (different buttons activated two screens)
[image]

I suspect I can find "view change" option here in future release?
[image]

unocelli, 2022-04-08:
I already update by github master branch, and I'm coming to update docker rc version too, witch version are you using?

AgrisNN, 2022-04-09:
My comment appeared a bit earlier, later I checkeed `master` branch and got update. 
Thanks for commiting the update.

unocelli, 2022-05-08:
Let me know if something doesn't work as expected

closed 2022-05-08 by unocelli

thewalk, 2022-05-14:
Is this full feature released with 1.16. I tried it, but the view not change every time.
[image]
[image]

unocelli, 2022-05-14:
Hi,
try to set 'Time Delay' to 0

thewalk, 2022-05-14:
it works, but only once, from the mainview to testview, and it could not go back from testview to mainview. does the start view configuration in layout affects?