frangoteam/FUXA #323: Partitially missing OPC UA values

AgrisNN, 2022-04-08:
Dear Fuxa team,

During OPC UA tags reading some of tags are not updated and remain empty.
The list of tags is around 45 items long. Only ~30 are updated, 11 are stuck.
[image]

The log contains messages "'ns=4;i=35' _startMonitor Error: expecting a valid session here" 
which are repeated continuously.

<details><summary>LOG File (click to expand)</summary>

```2022-04-07T19:35:50.295Z [error] 'ns=4;i=48' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.297Z [error] 'ns=4;i=35' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.298Z [error] 'ns=4;i=36' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.300Z [error] 'ns=4;i=34' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.301Z [error] 'ns=4;i=52' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.302Z [error] 'ns=4;i=56' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.303Z [error] 'ns=4;i=57' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.305Z [error] 'ns=4;i=60' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.339Z [error] 'ns=4;i=45' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.341Z [error] 'ns=4;i=46' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.342Z [error] 'ns=4;i=48' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.344Z [error] 'ns=4;i=35' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.345Z [error] 'ns=4;i=36' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.346Z [error] 'ns=4;i=34' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.348Z [error] 'ns=4;i=52' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.349Z [error] 'ns=4;i=56' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.350Z [error] 'ns=4;i=57' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.351Z [error] 'ns=4;i=60' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.381Z [error] 'ns=4;i=53' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.382Z [error] 'ns=4;i=33' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.384Z [error] 'ns=4;i=10' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.385Z [error] 'ns=4;i=11' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.386Z [error] 'ns=4;i=25' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.387Z [error] 'ns=4;i=26' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.389Z [error] 'ns=4;i=39' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.390Z [error] 'ns=4;i=40' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.391Z [error] 'ns=4;i=41' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.393Z [error] 'ns=4;i=42' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.394Z [error] 'ns=4;i=45' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.395Z [error] 'ns=4;i=46' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.399Z [error] 'ns=4;i=48' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.401Z [error] 'ns=4;i=35' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.402Z [error] 'ns=4;i=36' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.403Z [error] 'ns=4;i=34' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.405Z [error] 'ns=4;i=52' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.406Z [error] 'ns=4;i=56' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.408Z [error] 'ns=4;i=57' _startMonitor Error: expecting a valid session here
2022-04-07T19:35:50.409Z [error] 'ns=4;i=60' _startMonitor Error: expecting a valid session here
```
</details>

Few lines in error-log appears sometimes, not too often
```
2022-04-07T13:45:04.741Z [error] Error: expecting a valid session here
at assert (/home/pi/FUXA-1.1.5/server/node_modules/node-opcua-assert/dist/index.js:11:21)
at ClientSubscriptionImpl.get session [as session] (/home/pi/FUXA-1.1.5/server/node_modules/node-opcua-client/dist/private/client_subscription_impl.js:188:40)
at Function._toolbox_monitor (/home/pi/FUXA-1.1.5/server/node_modules/node-opcua-client/dist/client_monitored_item_toolbox.js:38:38)
at /home/pi/FUXA-1.1.5/server/node_modules/node-opcua-client/dist/private/client_subscription_impl.js:768:72
at Immediate.waitForSubscriptionAndMonitor (/home/pi/FUXA-1.1.5/server/node_modules/node-opcua-client/dist/private/client_subscription_impl.js:527:17)
at processImmediate (node:internal/timers:466:21)
```

The only workaround to get OPC tags is increasing polling time. 
For my 45 tags 2 seconds (sic!) is only an option to get all tags polled. 
Which slows down the UI and makes it unresponsible.

Any cleu?
Thanks in advance.

unocelli, 2022-04-08:
Hi,
can depend on several aspects: 
fuxa, opcua server hardware. which opcua server it is?

AgrisNN, 2022-04-09:
Hi, 

Fuxa is latest `master` updated few hours ago. 
OPC server is S7-1200 internal OPC running on PLC.
Same server polled by external client is working well, you may see how frequent the value is being changed
[image]

unocelli, 2022-04-09:
Have you try to refresh the table value with the button?
[image]

AgrisNN, 2022-04-09:
Oh yeah! One of my favorite buttons, but no cigar.

unocelli, 2022-04-09:
If you restart the fuxa server you have every time the same logs errors (of items)?

AgrisNN, 2022-04-09:
well I'd say no. How `fuxa.log` looks like after 5 minutes after restart - with no errors.

```
 System Logs
2022-04-07T19:21:21.143Z [info] FUXA V.1.1.5-666
2022-04-07T19:21:21.210Z [info] daqstorage: init successful!
2022-04-07T19:21:21.748Z [info] api: init successful!
2022-04-07T19:21:21.763Z [info] plugin-installed node-opcua 2.65.1
2022-04-07T19:21:21.766Z [info] plugin-installed node-bacnet 0.2.4
2022-04-07T19:21:21.767Z [info] plugin-installed node-snap7 1.0.6
2022-04-07T19:21:21.772Z [info] usrstorage.connected /_appdata/users.fuxap.db database
2022-04-07T19:21:21.774Z [info] prjstorage.connected-to /_appdata/project.fuxap.db database
2022-04-07T19:21:21.788Z [info] project.prjstorage-init-successful!
2022-04-07T19:21:21.790Z [info] runtime init project successful!
2022-04-07T19:21:21.795Z [info] users.usrstorage-init successful!
2022-04-07T19:21:21.797Z [info] runtime init users successful!
2022-04-07T19:21:22.012Z [info] plugin-installed modbus-serial 8.0.5
2022-04-07T19:21:22.014Z [info] runtime init plugins successful!
2022-04-07T19:21:22.016Z [info] FUXA init in 939ms.
2022-04-07T19:21:22.728Z [info] daqstorage reset!
2022-04-07T19:21:22.733Z [info] 'FUXA Server' data loaded (0)
2022-04-07T19:21:22.734Z [info] 'FUXA Server' created
2022-04-07T19:21:27.022Z [info] 'OPCUA_S71200' data loaded (43)
2022-04-07T19:21:27.026Z [info] 'OPCUA_S71200' created
2022-04-07T19:21:27.032Z [info] 'FUXA Server' start
2022-04-07T19:21:27.035Z [info] 'OPCUA_S71200' start
2022-04-07T19:21:27.038Z [info] alarms check start
2022-04-07T19:21:27.038Z [info] notificator check start
2022-04-07T19:21:27.040Z [info] scripts.start-checkstatus
2022-04-07T19:21:27.047Z [info] FUXA started!
2022-04-07T19:21:27.063Z [info] WebServer is running http://127.0.0.1:1881/
2022-04-07T19:21:27.066Z [info] daqstorage.bind-daq-data '_db/daq-data_0_20220407201949.db' database
2022-04-07T19:21:27.066Z [info] daqstorage.bind-daq-data '_db/daq-data_0_20220407202122.db' database
2022-04-07T19:21:27.068Z [info] daqstorage.bind-daq-map '_db/daq-map_0.db' database
2022-04-07T19:21:27.070Z [info] daqstorage.bind-daq-data '_db/daq-data_d_048aba00-d8574732_20220407201955.db' database
2022-04-07T19:21:27.070Z [info] daqstorage.bind-daq-data '_db/daq-data_d_048aba00-d8574732_20220407202127.db' database
2022-04-07T19:21:27.071Z [info] daqstorage.bind-daq-map '_db/daq-map_d_048aba00-d8574732.db' database
2022-04-07T19:21:27.113Z [info] daqstorage.check-to-archive-db-file '_db/daq-data_0_20220407201949.db'
2022-04-07T19:21:27.116Z [info] daqstorage.connected-to-map '_db/daq-map_0.db' database
2022-04-07T19:21:27.118Z [info] daqstorage.check-to-archive-db-file '_db/daq-data_d_048aba00-d8574732_20220407201955.db'
2022-04-07T19:21:27.184Z [info] daqstorage.connected-to-map '_db/daq-map_d_048aba00-d8574732.db' database
2022-04-07T19:21:27.187Z [info] daqstorage.connected-to-data '_db/daq-data_0_20220407202122.db' database
2022-04-07T19:21:27.188Z [info] daqstorage.connected-to-data '_db/daq-data_d_048aba00-d8574732_20220407202127.db' database
2022-04-07T19:21:27.195Z [info] daqstorage.load-map successful! '0'
2022-04-07T19:21:27.248Z [info] daqstorage.load-map successful! 'd_048aba00-d8574732'
2022-04-07T19:21:27.250Z [info] daqstorage.check-to-archive-db-file '_db/daq-data_0_20220407201949.db' database deleted!
2022-04-07T19:21:27.251Z [info] daqstorage.check-to-archive-db-file '_db/daq-data_d_048aba00-d8574732_20220407201955.db' database deleted!
2022-04-07T19:21:27.302Z [info] socket.io client connected
2022-04-07T19:21:27.780Z [info] 'OPCUA_S71200' connection
2022-04-07T19:21:27.836Z [info] socket.io client connected
2022-04-07T19:21:27.866Z [info] socket.io client connected
2022-04-07T19:21:28.004Z [info] 'OPCUA_S71200' connected!
2022-04-07T19:21:28.041Z [info] alarmsstorage.connected-to /_appdata/alarms.fuxap.db database.
2022-04-07T19:21:28.046Z [info] alarms.alarmstorage-init-successful!
2022-04-07T19:21:30.648Z [info] 'OPCUA_S71200' subscription created!
2022-04-07T19:21:30.663Z [info] socket.io client connected
2022-04-07T19:22:04.475Z [info] socket.io client connected
2022-04-07T19:22:12.251Z [info] socket.io client connected
```

Some items are still missing
[image]

I suppose errors begin to appear later, after some time.
Now I run a test - restarted Fuxa, and made no mouse clicks on server side. Just doing nothing to see if any error will appear in `fuxa.log`
I'll report the result in 15-20 minutes.

AgrisNN, 2022-04-09:
Hi again,

Well, 20 minutes after restart - no error messages appeared in log. Moreover, log is exactly the same as my previous post.
This leads to an idea: once you start to work with Connections in Editor (for example, edit polling time, open list of tags, etc) - Fuxa log file start to be filled with that: 
```
2022-04-07T20:07:03.438Z [info] 'OPCUA_S71200' stop
2022-04-07T20:07:03.747Z [warn] 'OPCUA_S71200' Warning => Session closed
2022-04-07T20:07:03.807Z [info] 'OPCUA_S71200' exist
2022-04-07T20:07:03.810Z [info] 'OPCUA_S71200' data loaded (43)
2022-04-07T20:07:03.812Z [info] 'OPCUA_S71200' start
2022-04-07T20:07:03.960Z [info] 'OPCUA_S71200' connection
2022-04-07T20:07:04.180Z [info] 'OPCUA_S71200' connected!
2022-04-07T20:07:04.273Z [info] 'OPCUA_S71200' subscription created!
2022-04-07T20:07:14.797Z [warn] alarms working (check) overload!
2022-04-07T20:07:53.711Z [info] 'OPCUA_S71200' session keepalive
2022-04-07T20:08:15.702Z [info] socket.io client connected
2022-04-07T20:08:15.714Z [info] socket.io client connected
2022-04-07T20:08:15.728Z [info] socket.io client connected
2022-04-07T20:08:15.745Z [info] socket.io client connected
2022-04-07T20:08:34.557Z [info] 'OPCUA_S71200' stop
2022-04-07T20:08:34.878Z [warn] 'OPCUA_S71200' Warning => Session closed
2022-04-07T20:08:34.950Z [info] 'OPCUA_S71200' exist
2022-04-07T20:08:34.954Z [info] 'OPCUA_S71200' data loaded (43)
2022-04-07T20:08:34.958Z [info] 'OPCUA_S71200' start
2022-04-07T20:08:35.100Z [info] 'OPCUA_S71200' connection
2022-04-07T20:08:35.306Z [info] 'OPCUA_S71200' connected!
2022-04-07T20:08:35.409Z [info] 'OPCUA_S71200' subscription created!

```
Three or four modifications made via user interface to Connections (I was changing polling time), and here we go:
`2022-04-07T19:35:50.297Z [error] 'ns=4;i=35' _startMonitor Error: expecting a valid session here`
repeated continuously.

As a conclusion: once you've made modification to Connection settings, TCP connection to OPC Server closes, and another connection opens again. Since S7-1200 PLC has a limitation to concurrent connections (as far as I remember, there are 4), Fuxa can't establish new connection (four previous connections stay in TIME_WAIT state I suppose and are not reistablished, but will die soon themselves inside PLC).
As a result we see `Error: expecting a valid session here` in log file, hundreds of them, until PLC releases dead connection and it can be established again - and since then no new error appear 'cause connection being made.

unocelli, 2022-04-09:
It is possible that the problem is the table view that have problems to
refresh the tags? I see that i added a pager that doesn't work, i have to
fix it. If you bind the tags to your gui (shapes) have you issues?

AgrisNN, 2022-04-09:
Well the answer is yes and no.

"Yes" answer: 
There's a problem with table refresh, right. Assigned shapes in View display values as they should be.

"No answer": 
the situation drastically depends on "polling time". 
If polling time is 500ms, NOTHING gets updated. No errors in log. It looks like no tags updates are coming, but connection stays green in Connections list.
If I change polling time to 2seconds (or 3 seconds), tags on shapes are updated, table in Editor not always. 

The polling time is critical, 2 seconds - too slow :(

AgrisNN, 2022-04-09:
(my big message I've been preparing whole day)

I suppose the part "error messages" of this issue is clarified now. 
There's nothing to do - the OPC Server inside PLC is the source of problem. 
Answer is: start your PLC, FUXA and do to touch anything :)

The initial question is UNREVEALED: why do we see partitially missing values on Opc Tags page?
This is not related to error messages discussed here  previously. 

I suspect the following behavior, and this has to be resolved somehow: 
- once PLC starts, it sets OPC tags to some values
- my PLC program runs idle and DOES NOT change tags
- FUXA starts, and subscribes to OPC Server. 
- (my suspect) since tags remain unchanged, OPC doesn't tell FUXA to update tags, and Fuxa displays empty fields.
- (the proof) If later I change value of a any tag inside PLC, all tags are updated and list of Tags in Connections become filled with values, completely. BTW, I do not press anything in FUXA, only changing value of tag in PLC.

My thought is: when FUXA establishes a connection to OPC, it must read ALL values first as they are, and make a subscription to OPC. Only in this case at start time we can see actual values of PLC...

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli