frangoteam/FUXA #324: Data archiving

robsori, 2022-04-12:
Hi Unocelli, 
I come back with a new observation on how to archive data. I noticed that there is only one archiving time, namely 60. In the database I noticed that the data is archived at an interval of 1 second. I think it would be useful to be able to set different archiving times for each tag. For example, pressure and flow values are usually archived every 1 sec, but temperature values should be archived more slowly.
[image]
From my tests the data is archived indefinitely without the possibility of automatic deletion. It would be useful to be able to choose a time period in which to keep the data. For example, it would be useful to keep the data only for 30 days. This time period should be configurable.
I noticed that both the alarm and the data archive can be deleted manually, but it would be useful to do this independently of the operator's action, automatically.
[image]

robsori, 2022-08-25:
Hi,
please tell me if this option of selective archiving of tags and on different time intervals is possible in Fuxa in the near future. A database backup option is also needed. Thanks.

unocelli, 2022-11-13:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-11-13 by unocelli