frangoteam/FUXA #326: Blinking actions for different values of the same tag not working

IonuDragomir1, 2022-04-13:
Hi,
I'm encountering a problem on multiple blinking actions for different values of the same tag. For example, I want a rectangle to blink red-gray for a value of TagWord1 and green-gray for another value of same tag, TagWord1.
[image]
For the above configuration, my rectangle is blinking only for value 8192 as you can see in the below video.
https://user-images.githubusercontent.com/58516894/163147373-edb7dc24-7b64-4992-a446-1bc0253e4ec5.mp4
If for green color I would use tag TestWord1 and for yellow I'd use another tag, let's say TestWord2, it would work perfect. Why isn't working if I use the same tag?

unocelli, 2022-04-13:
Hi, Thanks for the report.
The conflict is because there is a shape restore when the tag value is not in the range to blink.
If you configure on the same tag, the action is checked for each time it has been configured, in your case 2 times, so you will have that once activates the blink while the second does the restore, depending on the sequence

IonuDragomir1, 2022-04-14:
Thank you for replying!
I understand what did you write, but it's not ok how it is working. As you can see in the video sent in my initial post, when the word has the value 0, the shape is restored. In my opinion, the normal functionality is to blink green when TestWord1 has the value 4096 or blink yellow when TestWord1 has the value 8192. If TestWord1 has any other value, the shape should be restored.

unocelli, 2022-05-08:
Let me know if something doesn't work as expected

closed 2022-05-08 by unocelli