frangoteam/FUXA #329: Alarms

eyadalhanafi, 2022-04-20:
Hello,

1 - Please is it possible to have the alarm settings like: Enabled, check Interval, Min, Max, Time Delay linked with tags so that operator with authorization can change those parameters on run time.   to have the option for static configuration values from Alarms configuration or Dynamic by linking those parameters with Tags (internal or PLC address)
2 - Provide search options for alarm history , by time range, Tags, ack By, etc. 
3 - Print options for Alarms history

Thank you for your efforts.

eyadalhanafi, 2022-04-21:
Hello,
it seems there is misunderstanding here. The case is not related to bit mask.

unocelli, 2022-04-21:
I'm sorry, my mistake, 
I would appreciate if you could separate the 3 issues (features) so that they are clearer to manage.