frangoteam/FUXA #331: Angular build

balincagabrielalin, 2022-04-20:
Hi!

I think you installed sass module without --save flag because have trouble compiling it. Also I think you are using older version of sass module because the compiler is giving me errors here : 

```
 .my-header-filter /deep/ .mat-sort-header-button {
    display: block;
    text-align: left;
    margin-top: 5px;
}

.my-header-filter /deep/ .mat-sort-header-arrow {
    top: -12px;
    right: 20px;
}
```

In the file device-map.component.scss, I removed /deep/ and now its working.

Screenshots : 
[image]
[image]

unocelli, 2022-04-20:
Hi,
thanks for the report, /deep/ is deprecated and in any case the device-map.component don't need it anymore, so i removed too. 
Interesting that I don't receive the error, witch global angular cli version are you using?

balincagabrielalin, 2022-04-21:
This is the angular version :

[image]

unocelli, 2022-05-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-05-08 by unocelli