frangoteam/FUXA #335: Some programming problems

jcxjcx123, 2022-04-24:
Hello Sir, I want to rewrite and add some features, but I am not familiar with Node.js and angular.js, so I would like to ask you some questions.  For example, login popover title, I'm in...  \ client\src\app\login\login.component.htmllogin.com ponent. HTML, change in the title does not work, need to go to the main XXXXX. Changes in js,  However, the amount of main.xxxx.js code is too much, it is not convenient to change, could you please give me a solution

unocelli, 2022-04-24:
Hi,
if you are debugging then, it should be compiled automatically, while if you want to see it on production (localhost:1881) you have to: 
```
cd client
ng build --prod

```

AngelFernandezSobrino, 2022-04-26:
Angular is a typescript framework which works by compiling the project source is TS into a bundled, "optimized", and obscured JS source code, which is the code sent to the client. Main.xxxx.js is the output of the compiler, and the code executed in the client frontend. 

Follow the instructions in https://angular.io/start/start-deployment to learn about the framework and how to deploy the frontend, otherwise should be really difficult to change properly and consistently the frontend behavior.

closed 2022-10-09 by unocelli