frangoteam/FUXA #337: Connection error for Siemens s7 1212C

balincagabrielalin, 2022-04-28:
Hi!

I have a Siemens S7 1212C plc behind a TP-Link GSM router with static IP. I can acces it with WinCC and Kepware OPC Server but I can't do that with FUXA. Is there a settings or something that I am missing ?

unocelli, 2022-04-28:
Hi,
I suppose you want to connect with opc ua? it could be a problem with permissions or certificate. have you tried removing them on the server? what's on the log file?

balincagabrielalin, 2022-04-28:
Sorry, I forgot to specify, I'm using the siemens driver.

unocelli, 2022-04-28:
you mean S7 protocol? I'm using [node-snap7](https://github.com/mathiask88/node-snap7) driver

balincagabrielalin, 2022-04-28:
I re-edited my original post. This problem is only for this type of plc (siemens s7 1212c) with newer ones, works fine.

closed 2023-05-14 by unocelli