frangoteam/FUXA #349: Empty script crashes nodejs

balincagabrielalin, 2022-05-05:
Hi!

If we make an empty script and make it run let's say every 1 sec that will crash the nodejs instance. 

The problem come's from file server/rntime/scripts/index.js line 115  where there is a reject (as far as I know this is used only for promises)

```
/**
     * Check the Scripts state machine
     */
    var _checkStatus = function () {
        if (status === ScriptsStatusEnum.INIT) {
            if (_checkWorking(true)) {
                _init().then(function () {
                    status = ScriptsStatusEnum.LOAD;
                    _checkWorking(false);
                }).catch(function (err) {
                    _checkWorking(false);
                });
            }
        } else if (status === ScriptsStatusEnum.LOAD) {
            if (_checkWorking(true)) {
                _loadProperty().then(function () {
                    _checkWorking(false);
                    status = ScriptsStatusEnum.IDLE;
                }).catch(function (err) {
                    _checkWorking(false);
                });
            }
        } else if (status === ScriptsStatusEnum.IDLE) {
            const time = new Date().getTime();
            Object.keys(schedulingMap).forEach((name) => {
                const script = schedulingMap[name];
                if (script.isToRun(time)) {
                    try {
                        scriptModule.runScriptWithoutParameter(script);
                        script.lastRun = time;
                    } catch (err) {
                        reject(err);                <---- This is the problem
                    }
                }
            });
        }
    }
```

unocelli, 2022-05-05:
Hi,
Thanks for the report, I will fix it.

unocelli, 2022-07-26:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-26 by unocelli