frangoteam/FUXA #352: Adding a custom alarm control

IonuDragomir1, 2022-05-09:
Hello,

It'd be useful if an user could add an alarm control on a certain page. I know that FUXA has the alarm feature which is available for the entire project. The alarm control I propose is meant to display only the alarms of a certain mechanism. For example, in the below image, I'm showing you, from WinCC, a window (faceplate) which is used to control and monitor a pump. This window has multiple tabs. One of it contains an alarm control which displays only pump's alarms. 
[image]

Kiruthick-Roxx, 2024-09-17:
Hi,
Any updates on this?

unocelli, 2024-10-23:
Hi, is available in the latest version (Table control), let me know if you have any issues.

closed 2024-10-27 by unocelli