frangoteam/FUXA #356: Controlling Alarm Properties of Tags and Groups at Run Time

eyadalhanafi, 2022-05-10:
Thank you for your efforts.
I would like to know if possible one can monitor and control a specific tag’s alarm information on runtime.
The concept is common on most of tag based SCADA systems.

suggestions is to have something called dotfields associated with a tag, use this syntax: tag.dotfield to access (read/write) the data

Example:
If you want to allow run-time changes to the Hi alarm limit on a tag named Analog_tag, you can create an Analog – User Input touch link to a button and enter Analog_tag.HiLimit as the expression in the link’s dialog box. During run time, the operator simply clicks the button and types in a new value for the Hi alarm limit assigned to the Analog_tag.

Here it sample of dotfields associated with alarm tag.
.Ack
Acked
.Alarm
.AlarmEnabled
.AlarmDelay
.HiHiLimit
.HiLimit
.LoLimit
,LoLoLimit
.HiStatus
.LowStatus
.AlarmDesc

unocelli, 2022-05-13:
Hi,
you mean to be able to receive the alarm object with its attributes in the script and be able to change its value?

eyadalhanafi, 2022-05-13:
Yes that's right but it is not necessary to receive and write values using a script. using script is one option.
Use internal tags for example to link with alarm object attributes. the most important thing is to be able to change those value on runtime.