frangoteam/FUXA #360: Can I call a get request on another web server by FUXA Script?

Pichitor, 2022-05-15:
Something like this:
const Http = new XMLHttpRequest();
const url='https://jsonplaceholder.typicode.com/posts';
Http.open("GET", url);
Http.send();

AngelFernandezSobrino, 2022-05-15:
The script is executed in the nodejs server, not in the client, XHR is not natively supported on nodejs, it can only be used in browsers, or adding a module that implements it. You have to use native nodejs implementations such as Http module or add a custom dependency.

unocelli, 2022-10-09:
Hi, Recently @mauroalexandre added the possibility to run a script in frontend. did you try?

unocelli, 2023-05-14:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-05-14 by unocelli