frangoteam/FUXA #361: Modbus TCP Plugin is not working

eyadalhanafi, 2022-05-15:
Hi.
It seems there is an issue in Modbus TCP plugin.
It doesn't read data from slave device.
I used Modsim32 to simulate slave device.

I'm using Fuxa Version 1.1.6-688
Node.js v16.15.0.

[image]
[image]

unocelli, 2022-05-17:
Hi,
I would try with Node.js v14?

eyadalhanafi, 2022-05-18:
Hi,
I tried with Node.js Version 14 and it is the same. You can give it a try as well.

unocelli, 2022-05-18:
I just tested with ModRSsim2, in ubuntu, windows and raspberry, work fine!

try to remove and reinstall node_modules

eyadalhanafi, 2022-05-19:
I removed Fuxa and reinstall it again but it is the same. Any idea what could be the issue?

unocelli, 2022-05-20:
something is still not clear to me:
old versions worked and with the new one it no longer works?
have you tried another modbus simulator (ModRSsim2)?

Rainproof, 2022-05-21:
Hi,
I have FUXA installed on my Raspberry Pi.
In the attached video, the connection with modbus poll is modbus tcp.
I got anxious about the new version of FUXA and decided to check it out.
[image]

eyadalhanafi, 2022-05-23:
Old version of Fuxa, Modbus was working smoothly. This happened when i installed the latest version.
I will try with reinstall old version and check again.

eyadalhanafi, 2022-06-16:
I tested also on version Version 1.1.6-727 but it is not working also.

closed 2022-11-13 by unocelli