frangoteam/FUXA #363: Error while installing fuxa

rslord8, 2022-05-23:
Hello. Error while installing fuxa
[image]
Tell me how can I fix it?

rslord8, 2022-05-23:
please tell me how to completely remove FUXA?
I want to reinstall FUXA.

unocelli, 2022-05-23:
npm uninstall -g --unsafe-perm @frangoteam/fuxa

rslord8, 2022-05-24:
an error occurs during reinstallation.

And also tested on the docker, polling via Modbus TCP, OPC UA does not work.

[image]

[image]

unocelli, 2022-06-14:
I saw that you added feature requests and wondered if this issue was resolved?

rslord8, 2022-06-17:
Reinstalled docker and FUXA on docker. Modbus TCP does not interrogate the slave device.

[image]
[image]
[image]

unocelli, 2023-05-14:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-05-14 by unocelli

rslord8, 2023-05-22:
Hello,
Thanks, everything is working now.