frangoteam/FUXA #370: MQTT subscription issue

jcxjcx123, 2022-05-25:
Hi!
When I subscribe to a topic using MQTT, I find that the parsed content is an object, which is actually a string. Is there a problem with the data format?
[image]
[image]

unocelli, 2022-05-25:
Hi, the problem is the array...currently is not supported. there is a work around that is a bit tricky: you can make the subscription as a raw (string) and then make a script that converts and assigns to tags on the FUXA server

Schrolli91, 2022-06-08:
@unocelli how can such a script looks like?

Schrolli91, 2023-04-26:
still need a feature to get data from inside. how can your mentioned script looks like? @unocelli

unocelli, 2023-04-26:
Hi, you have to define the topic
[image]

and in script you can ask
[image]

jingshui127, 2023-07-25:
Is there a manual for the script?

unocelli, 2023-07-25:
Hi, we have nothing yet. you can try in [discord ](https://discord.com/channels/1045638823113138206/1105054356887322674)

catalinzeb, 2024-02-19:
i tried this and it gives me a 400 Bad Request error

unocelli, 2024-02-23:
@catalinzeb you have probably not configured correctly