frangoteam/FUXA #385: Is it possible to add X-Y Graph?

top58525785, 2022-06-05:
HI Fuxa team:

Is it possible to add X-Y Graph Chart Since some  X Axis data is not usually always Time?

We can define Chart X Axis Data and Y Axis Data, Such as array

Thanks

unocelli, 2022-06-06:
Hi,
i disagree with you. in a scada system or where you are operating with sensors, **usually** you have time series charts. 
I would be interested to know what kind of purpose you need a chart like this for? 
It would help me to have an example of how to configure and bind tag values with X-Y axes.

superkalt, 2022-06-06:
I think the graphic chart should only show a short period of time in a SCADA. There is much better software on the market for this. Influxdb & Grafna does this job brilliantly.

My way:
Bring all information to the MQTT broker
Import all data into an influxDB
create the appropriate representations with Grafana

in a SCADA it's great to see how a PID controller behaves! But that happens in seconds or minutes, and time always plays the important role

top58525785, 2022-06-06:
for example normally we get two sensor data at the same time，then we need build an chart graph with such data A to be x axie and data B to be Y axie。
such as data B relations with data A…

发自我的iPhone

unocelli, 2024-02-23:
Hi, we have just added this feature, it would be nice to have your feedback

unocelli, 2024-05-22:
is available in the latest version,  let me know if you have any issues.

closed 2024-05-22 by unocelli