frangoteam/FUXA #386: error comes up when  run test for Script

top58525785, 2022-06-05:
HI fuxa team:
i meet a error when using run test for script

 >Http failure response for /api/runscript: 400 Bad Request

How to fix?
Thanks

unocelli, 2022-06-06:
Hi,
if you share your script maybe I can help you, in any case you can use console.log to debug

top58525785, 2022-06-07:
HI FUXA TEAM:
i write as what you example show , also it doesn't work
I use Internal Device Tag to do
Please check the attachment Picture
[image]

unocelli, 2022-06-07:
internal device is only for frontend function, the flow don't go to backend, I have to remove it from the choose, please use FUXA Server tags

top58525785, 2022-06-07:
Thanks for reply
But my question is that if we use internal device tag，we need to some simulation data with script， it
will not support？
Jason

发自我的iPhone

unocelli, 2022-06-08:
No, isn't supported, you can use FUXA Server tag to make simulation
[image]

top58525785, 2022-06-09:
HI unocelli
Thanks for you reply asap
I did what as you said, the scipt run is no response
please check the attachment
[image]
[image]

top58525785, 2022-06-09:
does fuxa tag support script？

发自我的iPhone

unocelli, 2022-06-09:
if you do not receive the console message, then the problem is not the script
[image]

SNoble6, 2022-06-17:
Is there a way to do:
`import {Buffer} from 'node:buffer';`
?

unocelli, 2022-07-24:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-24 by unocelli