frangoteam/FUXA #388: Submenu for sidenav

unocelli, 2022-06-05:
### Discussed in https://github.com/frangoteam/FUXA/discussions/312

<div type='discussions-op-text'>

<sup>Originally posted by **AlexandruAndrei3** March 28, 2022</sup>
Hi,
When adding multiple views and tabs the home side navigation tends to get pretty crowded. For a better structure I would suggest that it would be useful to offer the user the option to add submenu items in the layout settings something similar to a dropdown but with the same style set up in the Navigation Side Menu Type. 
[image]

Thanks, for your time!</div>

liulujian00, 2023-11-15:
I think this is a good design for the interface layout. I am currently encountering the problem of insufficient tab classification. This submenu can just solve my problem. I would like to ask, how is the development of this function and when will it be released?

unocelli, 2025-12-26:
Available in the latest version