frangoteam/FUXA #394: InternalVariables pliugin is missing

GTB4, 2022-06-07:
Hi, 

I have an issue with the internal variables. When i try to create an internal variable i do have an error on my terminal : [WAR] try to create InternalVariables but plugin is missing and when i try to write a value on the variable the value of the var don't change.

 
[image]
[image]

Thanks.

unocelli, 2022-06-07:
internal device is only for frontend function, the flow don't go to backend, I have to remove it from the choose, please use FUXA Server tags

GTB4, 2022-06-08:
Thanks a lot for you reply !

unocelli, 2022-07-24:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-24 by unocelli