frangoteam/FUXA #399: Where are the plugins when installing with npm?

dreamersp, 2022-06-08:
Greetings,
Congratulations and thanks for the great work and effort put in this project.
I can't find the plug-ins referenced in the documentation and README like the ones of MQTT, Modbus, etc,  when installing with `npm i @frangoteam/fuxa` in a project, the message `2022-06-08T17:05:15.950Z [ERR]  runtime.failed-to-init plugins` appears and none appear in the editor.
I must be missing something or maybe could be explained how to obtain them in the documentation?

Thanks, KR.

unocelli, 2022-06-08:
Hi,
you don't need to install MQTT plugin, for Modbus look in [wiki](https://github.com/frangoteam/FUXA/wiki/HowTo-Devices-and-Tags). If is not workin try to restart the server or try to install manually #284

unocelli, 2022-07-29:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-29 by unocelli