frangoteam/FUXA #401: Boolean type problems

jcxjcx123, 2022-06-09:
Hi,
I seem to be running into some problems. My current Boolean data is of true/false type, but it seems to bind only 0/1 in fuxa system, so what should I do? Thank you very much for your help.
[image]
[image]

unocelli, 2022-06-09:
Hi, I suppose you have try :wink:
0=false 
1=true

unocelli, 2022-06-10:
FUXA make the convertion from boolean to number before process the logic.

jcxjcx123, 2022-06-10:
But when I use the Led Gauge to bind a bool and add blinking events, the Led Gauge does not change. Am I setting it wrong
[image]

unocelli, 2022-06-10:
I seem to see that your tag changes value quickly, each time it goes to 0 the blink is reset and the led returns to its initial state, so the blink animation does not start in time

jcxjcx123, 2022-06-10:
I thought that was the reason at first, but when I increased the polling time to 5 seconds, the result was still the same, so I wanted to consult you

unocelli, 2022-06-10:
I don't think that change the device polling is the solution...
[image]

jcxjcx123, 2022-06-10:
Well, thank you very much for your help. I will find the reason again

unocelli, 2022-06-10:
let me know

jcxjcx123, 2022-06-10:
I've done a lot of testing and I guess it might be the data type that causes this problem. I noticed that all the data types that subscribe via JSON are JSON. What's your opinion？
[image]
[image]

unocelli, 2022-06-10:
Yes, you are right, I can reproduce it. I have to fix it...thank you very much

jcxjcx123, 2022-06-13:
Hi,
Has this problem been fixed? I updated version V.1.1.6-727, but the same problem still seems to exist ...
[image]

unocelli, 2022-06-13:
Hi,
this is another problem, not so critical but to be fixed, tags values are only sent to the frontend if there is a change, change of view or at start-up, and animations are only checked if the value of the respective tag has had a change. if you do an F5 refresh it should work as expected.
about the change of color of the border: I can't explain it, let me know if you can reproduce it.

unocelli, 2022-07-26:
going to close this as resolved. let me know if you have any issues.

closed 2022-07-26 by unocelli