frangoteam/FUXA #404: the falshing tags

serite, 2022-06-12:
when I sub a topic mqttclient json object, the json item count over 20,the device tags allway falshing,
where I connect the outtxt with the tags(any one of them) on the view ,the outtxt allway show the mqtt value and "NaN".
[image]
[image]

unocelli, 2022-06-14:
Hi,
I tested with over 50 json items in my local mosquitto broker without problem. can you try to investigate so that I can reproduce the problem?

superkalt, 2022-06-14:
[image]

superkalt, 2022-06-14:
https://user-images.githubusercontent.com/44277174/173689973-dbbbbf86-d55d-4ccd-b1da-fc0b6113d71f.mp4

superkalt, 2022-06-14:
sorry i have to update the docker container

serite, 2022-06-19:
I am rebuild the device tags,until the value not c.
step1:sub a mqttclint json msg.(over 20 item)
step2:make all item to tags .
step3:save project ,the tags value always blank ,a few secends then the values appear.
      if the values always flash (maybe depend on the device interval setting). the I delete all tags on this device,
      redo all the steps; until the values not flash . 
      
maybe the project save operator tigger the bugs. same times ,i redo it over five times.

serite, 2022-06-19:
I am rebuild the device tags,until the value not flashing.
step1:sub a mqttclint json msg.(over 20 item, it is always happen)
step2:make all item to tags .
step3:save project ,the tags value always blank ,a few secends then the values appear.
      if the values always flash (maybe depend on the device interval setting). the I delete all tags on this device,
      redo all the steps; until the values not flash . 
      
maybe the project save operator tigger the bugs. same times ,i redo it over five times.

serite, 2022-06-19:
I am rebuild the device tags,until the value not flashing.
step1:sub a mqttclint json msg.(over 20 item, it is always happen)
step2:make all item to tags .
step3:save project ,the tags value always blank ,a few secends then the values appear.
      if the values always flash (maybe depend on the device interval setting). the I delete all tags on this device,
      redo all the steps; until the values not flash . 
      
maybe the project save operator tigger the bugs. some times ,i redo it over five times.

serite, 2022-06-19:
I used the newest docker fusa.
I am rebuild the device tags,until the value not flashing.
step1:sub a mqttclint json msg.(over 20 item, it is always happen)
step2:make all item to tags .
step3:save project ,the tags value always blank ,a few secends then the values appear.
      if the values always flash (maybe depend on the device interval setting). the I delete all tags on this device,
      redo all the steps; until the values not flash . 
      
maybe the project save operator tigger the bugs. some times ,i redo it over five times.

serite, 2022-06-19:
in the dev_3 (devices settings). the fusa load err prj ,it happend again. the ok prj is ok!
[Uploading prj.zip…]

serite, 2022-06-19:
[prj.zip](https://github.com/frangoteam/FUXA/files/8935773/prj.zip)
in the dev_3 (devices settings). the fusa load err prj ,it happend again. the ok prj is ok!

ouinouin, 2022-06-25:
Hi, 
I also got this kind of issues with MQTT ans json lists. (i to test FUXA I use json output from tasmota devices)

unocelli, 2022-06-30:
Hi,
in mqtt fuxa make subscription on the topics and will receive a callback with values that will be buffered, the polling look to the buffer and send to the frontend. I propose you to add a console.log to look what will be received in 'server/runtime/devices/mqtt/index.js line 339:
`console.log(topicAddr + ' = ' + msg);`

serite, 2022-07-03:
I add the console.log("...") into index.js ,when i pub the topic info ,as the same time the server log  flash . but the web is still flashing per sec and the server log not contine . and in the chrome console is not any log been created.

top58525785, 2022-07-04:
hi Serite , about the flashing issue about mqtt problem , are there any fix ways?

serite, 2022-07-06:
> hi Serite , about the flashing issue about mqtt problem , are there any fix ways?
restart the docker ,it will ok .but  if you modify same tags , it maybe happen again.

unocelli, 2022-07-07:
I am looking but I cannot reproduce it

ouinouin, 2022-07-08:
@unocelli  I didnt have a look yet, but could this issue have a link with the retain parameter of the mqtt  ? are you doing a new subscribe each time you poll the data ? or is the mqtt communication running continuously and just buffers the latest data available ?

serite, 2022-07-10:
http://121.36.211.235:1881/device   or http://121.36.211.235:1881/home  , you can test it any time.  and sub a topic info on the mosquitto in this ip .

ouinouin, 2022-07-17:
hi, to add some infos even of low value... i tried to add the log line 339 ( console.log(topicAddr + ' = ' + msg);) then closed fuxa then rebooted, then i couldnt reproduce this bug.... i deleted the debug line , and the bug didnt reappear... dont know what to think about that.

serite, 2022-07-19:
this is the way to deal this problem by me too. rrestart fuxa server .the bug is gone until the tags be modifed or rebuilded and save the project (not every time, meybe 30~50%).  i think meybe the buffer been affected but the souce data is ok .

ZhuNathan, 2022-10-07:
I test this issue with v1.1.8, it will be normal after I creat the new mqtt tags, but the issue will happen after I click "Save Project", it just an appearance what I see, I didn't find the root, I hope it will be helpful for this issue fixing.
same issue as #414 
https://user-images.githubusercontent.com/49235335/194551536-e7553049-bfd5-480f-b8ea-328c15204308.mp4

unocelli, 2022-10-12:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-10-12 by unocelli