frangoteam/FUXA #412: History charts problem

jcxjcx123, 2022-06-21:
Hello,
I would like to ask how the history chart is set up. At present, the real-time chart I set has data, but the history chart is empty, and they have the same data source.
The top two are real-time charts and the bottom two are history charts.
Thanks.
[image]

unocelli, 2022-06-21:
Hi,
have you configured the tags for the history? default is off
[image]

jcxjcx123, 2022-06-22:
I tried it. I had the tag turned on, but it didn't seem to work. Even the real-time chart was empty.
[image]