frangoteam/FUXA #414: mqtt issue

top58525785, 2022-06-23:
we have stop publish data to mqtt server long time
but fuxa is not stopping refreshing data
please check the picture and vedio
[image]

https://user-images.githubusercontent.com/54358179/175235905-745759b9-4fcd-4a60-85e3-e37ca066c500.mp4

https://user-images.githubusercontent.com/54358179/175235925-7a8f46e7-b84c-43cf-a91f-14444f984993.mp4

unocelli, 2022-06-25:
Hi,
is it possible that the problem is that FUXA ask the data to the broker who responds with the value of the last publish?

top58525785, 2022-06-25:
maybe，it is the problem for fuxa mqtt loop data acquire time seettings？
so how to fix it？

发自我的iPhone

unocelli, 2022-06-25:
maybe I am misinterpreting the mqtt protocol, but if a sensor only publishes data if it has changes or every minute and for example fuxa process the display or script with logic in a 1-second interval, what value should it take?

top58525785, 2022-06-26:
Actually fuxa should update and dispaly each data from the mqtt broker subscriber，not only the last data？ it should be the real time data from the broker..you can have a test for mqtt if sensor data publish data faster..

发自我的iPhone

superkalt, 2022-06-26:
this is a general question. Does fuxa also receive data when no user is logged on?
I see fuxa as pure visualization. it is therefore only necessary to receive and process data when a user is logged on.
MQTT has the "retain" function. If "Retain=true" the broker saves the last data transfer and when a new "client" logs in and subscribes to a "topic" he gets this last value. if "retain=false" it will not get any data until new data is sent again.

unocelli, 2022-06-26:
@superkalt I like your proposal, adding a configurable timeout, default 60 seconds:
- timeout < orange < 5 * timeout
- 5 * timeout < grey < 10 * timeout

top58525785, 2022-06-27:
So Unocelli:
How to fix the problem as vedio shows
IT display realdata and then zero data?

superkalt, 2022-06-27:
> So Unocelli: How to fix the problem as vedio shows IT display realdata and then zero data?

hmmm, come down.

superkalt, 2022-06-27:
> @superkalt I like your proposal, adding a configurable timeout, default 60 seconds:
> 
> * timeout < orange < 5 * timeout
> * 5 * timeout < grey < 10 * timeout

perhaps a good time to separate the processing of the values from the visualization?

serite, 2022-07-03:
this problem is same as  #404

top58525785, 2022-07-04:
@serite  is the problem fixed?

jcxjcx123, 2022-07-04:
I had this problem before, but I pulled the latest version and the problem was gone

unocelli, 2022-07-07:
I am looking but I cannot reproduce it

ZhuNathan, 2022-10-07:
I test this issue with v1.1.8, it will be normal after I creat the new mqtt tags, but the issue will happen after I click "Save Project", it just an appearance what I see, I didn't find the root, I hope it will be helpful for this issue fixing.
https://user-images.githubusercontent.com/49235335/194548008-0b5ea247-c768-4886-8aac-f32897bb142d.mp4

unocelli, 2022-10-08:
Hi, 
Thanks for the report, so I could reproduce it and I think I fixed it, can you confirm me?

ZhuNathan, 2022-10-08:
> Hi, Thanks for the report, so I could reproduce it and I think I fixed it, can you confirm me?

I test this in docker, it worked normally, that's wonderful!

unocelli, 2022-10-12:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-10-12 by unocelli