frangoteam/FUXA #415: Current Logged In User

eyadalhanafi, 2022-06-24:
Can we have the current logged in user as a tag to be displayed on the page?

superkalt, 2022-06-24:
let's discuss that first. this is a wish and not a misconduct

superkalt, 2022-06-24:
#416 
please close this Issue

eyadalhanafi, 2022-06-24:
Thank you.
Sorry but I don't think you understand my request.

superkalt, 2022-06-24:
i understand your request! 
a tag for each loggged user =>

superkalt, 2022-06-24:
at the moment two users can be logged in at the same time
if one of the user changes values, the other user wonders what just happens

superkalt, 2022-06-24:
> I'm sorry but that was not our request.

issues => is a problem of an already existing function
request => This is a request for a new feature

and new functions should first be discussed, maybe sit down at the computer and make a new branch, build in your function, and create a pull

concept10, 2022-06-26:
Does FUXA have the idea of logged-in users OR user permissions on objects on pages?

unocelli, 2022-06-30:
in fuxa you can enable user authentication, set show/enable authorisation per category to almost objects and navigation side menu, the authentication access is placed on the header bar right (the only place where I would go to search)

jingshui127, 2023-07-01:
Can I view the login history?

unocelli, 2023-07-01:
Hi,
only in Pro version

Jue07, 2024-02-29:
> in fuxa you can enable user authentication, set show/enable authorisation per category to almost objects and navigation side menu, the authentication access is placed on the header bar right (the only place where I would go to search)

Hi..why not log in user still can access the header items? even though it already shows/enabled for selected users only?