frangoteam/FUXA #423: Custom images to documentation

burner-, 2022-07-16:
Can you add to documentation how to add custom images. I see Images menu but I dont find way to upload those.

unocelli, 2022-07-29:
Hi,
what do you mean? where would you like to add your custom image?

ouinouin, 2022-10-14:
Hi,
@unocelli  i think i m in the same situation, 
 in the resources menu, there is a menu to add some supposedly previously imported images.
for myself i use docker container so the images are supposedly in : /usr/src/app/FUXA/server/_images
i ve tried to put svg and PNG images in this folder to check if i can see them in this section, without any results.
[image]

I suppose this menu is made to import some images in the graphics.

here my docker compose

`version: '3.3'
services:
    fuxa:
        ports:
            - '1881:1881'
        volumes:
            - './fuxa_appdata:/usr/src/app/FUXA/server/_appdata'
            - './fuxa_db:/usr/src/app/FUXA/server/_db'
            - './fuxa_logs:/usr/src/app/FUXA/server/_logs'
            - './fuxa_images:/usr/src/app/FUXA/server/_images'
        image: 'frangoteam/fuxa:latest'
`

cicci44, 2022-10-14:
.... theres no issue with the image/resources section..
just upload a folder within **_images** with a name and the svg inside that folder...
[image]

unocelli, 2023-10-10:
I’m going to close this as resolved. let me know if you have any issues.