frangoteam/FUXA #426: BACnet reads duplicated values

bencahill, 2022-07-20:
Hi,

This is really great software you have made. Thank you for all the time and effort you have put into it.

I had good success with reading Modbus values, but I am running into a problem with BACnet. When I add BACnet points, they all read as the same value as the first point added. See the attached screencast.

[image]

Any ideas how to fix this?

Thanks.

unocelli, 2022-07-29:
Hi,
does the problem always occur with any type of tags?

bencahill, 2022-07-29:
Yes, I just tested again with two binary inputs (BI) and two analog outputs (AO) and the same problem occurred as described above and illustrated in the screencast.

I also reproduced the problem with the YABE room simulator as mentioned in another issue ( https://github.com/frangoteam/FUXA/issues/114#issuecomment-803647744 ).

You can reproduce with a Windows box with YABE installed by running "C:\Program Files\Yabe\AddOn\Bacnet.Room.Simulator.exe". Look at points Temperature.Outdoor and Temperature.Indoor. They should be different.

bencahill, 2023-01-31:
Thanks. This is working now when I test it against the YABE simulator. Unfortunately I cannot get BACnet to work with any other device I have. Will add a note to #652.

closed 2023-01-31 by bencahill