frangoteam/FUXA #431: Input gauge is not updated when Tag value changes

algismo, 2022-07-28:
Hi,

I noticed that Input gauge ("Input value" control) gets its value fetched from a device only once during startup. If value changes in the device at a later time, gauge is not updated.

I wonder if it is a "feature" or a "bug"?

I am on the latest version (v1.1.7).

Thanks,
Algirdas

algismo, 2022-07-28:
I found that the affecting change was done by will (??) in this commit:
[image]

Can't come up with any good thoughts why it is made like that. It makes Input control little usable. If for example there are multiple scada instances where operators may change the same value, then one of them will be quickly out of sync. Also it is very weak safety if operator is adjusting a value that is not representing a true value that a PLC holds.

Any other thoughts?

unocelli, 2022-07-29:
Hi,
I don't know which is worse, as I am entering the value into the input control, the value is modified by the device :disappointed:

algismo, 2022-07-30:
To overcome this (modification of input value while entering a new value) one could use two strategies:
1. Enable input dialog. This is a very good feature. In such case operator has full control and the entered value is not being affected. Other scada systems even show live value beside the new one in such dialog to give even more overview.
2. Disable update of an input field while it has focus. In such case input field is being kept up to date to the tag value, but updates are "paused" while operator is writing in a new value.

I have some experience with FactoryLink Monitor Pro and Schneider Citect and both keep input files up to date and then the programmer of the system choose a strategy for modification (in place or through a dialog). In any way, from my point of view it is important to keep input fields up date. If scada system is restarted very rarely (normal) then operator never can be sure, that settings/inputs that he sees on the screen of his control system are actually in use or not. That is bad!

At least for my project this limitation is a show stopper. I started making necessary changes in my own fork but it would be nice to have more flexibility in the main project :)

Regards,
Algirdas

unocelli, 2022-07-31:
Thank you for your explanation, I like it.
I think the one to check the focus of the input control before modifying it is the easiest to apply. do you already have the modification that we can merge?

algismo, 2022-08-02:
I have the changes. As I changed more parts of the "input" gauge for my own needs, I will have to clean it up a bit before making the pull request.

I will come up with some more proposals for the "input" later 😃.

gentooza, 2022-09-06:
Hi @algismo ,
and sorry @unocelli for hijacking the issue, I see no other way to tak to @algismo, my message can be removed later.

I've some problems with the old factorylink for a university work and find nothing in the web, as you seem to have experience in factorylink software, could we speak privately? pleeease
my email is joa.cuellar@riseup.net

cheers, and sorry. (imagine how desperate I am XD)

closed 2023-02-05 by unocelli