frangoteam/FUXA #432: Schneider Electric/Modicon Premium PLCs. Support double word reading via Modbus

algismo, 2022-07-28:
Hi,

I am connecting FUXA to an old Schneider Electric/Modicon Premium PLC and reading tags using Modbus TCP. In those PLCs all numbers that are larger than one word (2 bytes) have their words swapped.
So to read large floats, I have to choose type Float32MLE. Same also applies to large integers (Int32 or so called Double Word (%MD)), but there is not such type as Int32MLE in FUXA. 

It is an easy change, similar to the one done in this commit: https://github.com/frangoteam/FUXA/commit/db6f9d99252418f16178ac0fa9f0ae3f45fff57e

I propose to introduce two new types: Int32MLE and UInt32MLE. 

By the way, does anybody know what "MLE" stands for?

Regards,
Algirdas

TomasAdvance, 2022-08-04:
Hi,
Same topic was discussed here
https://github.com/frangoteam/FUXA/discussions/275

algismo, 2022-08-04:
Thanks Tomas!

now it is clear ... It was a good explanation in the named ticket:
- Float - Big Endian (ABCD)
- Float - Little Endian (DCBA)
- Float - Mid-Big Endian (BADC)
- Float - **Mid-Little Endian** (CDAB) 

Actually it is described here as well: https://en.wikipedia.org/wiki/Endianness#Middle-endian . Was not patient enough to read entire article.

So by adding MLE for integers we will have a full coverage of the 32bit size.

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli