frangoteam/FUXA #435: mqtt broker topic to subscribe problem

guang067, 2022-08-01:
Dear teams:

     i am try the 1.1.7 version,   node version is 12.6.3,  and npm version is 6.14.4。 

     Devices setting seems fine:
[image]

but when add device tags and try to browser Topics on broker, no result found.   are there any thing wrong?    
[image]

public works ok,  and i use rabbitmq btw, thanks!
[image]

[image]

acgiesbrecht, 2022-08-05:
you need to add a # at the end of your topic stirng to browse al child topics: eg: iot/gw/device/#

guang067, 2022-08-25:
> you need to add a # at the end of your topic stirng to browse al child topics: eg: iot/gw/device/#

have try but still not work, also just use # to search all topic,  is sth wrong in rabbitmq? 
[image]

[image]

[image]

[image]

[image]

[image]

and the logs shows:
[image]

unocelli, 2022-09-15:
Hi,
I suppose the problem is not solved.
what happens if you try to discover an existing topic address?

guang067, 2022-09-22:
> 
thanks for reply,    and attached is the screenshot of this problem.  

https://user-images.githubusercontent.com/6702824/191687321-c87cecc4-8bbf-4563-ba75-f01983316e4e.mp4

unocelli, 2022-09-26:
something is missing

unocelli, 2022-09-29:
sorry my mistake...
have you already tried to test mqtt of FUXA by connecting to another broken like mosquitto?

jyb1975, 2022-12-01:
I'm facing the same problem...

jyb1975, 2022-12-12:
> I'm facing the same problem...
i got the reason. FUXA is using topic "#" to browse all topic on broken. However, FUXA receive nothing to some topics while those topics don't retain last value. It seems fine for MQTT5.0 broken in case of setting RETAIN flag of topic (last value will be retained on broken). RabittMQ supports MQTT3.1.1 only which RETAIN flag seems not working. 

one option solution, producer can send value to topic in little interval, say 1 second. FUXA should be working. 

anyway, is it possible to add topics on broken to input topic as well?

unocelli, 2023-03-19:
I am going to close this as resolved, let us know if you have any issues.

closed 2023-03-19 by unocelli