frangoteam/FUXA #437: Display the time and date

Teguhgithub, 2022-08-03:
How to display the time and date on FUXA web scada?

algismo, 2022-08-04:
I have a similar need as well. It would be a great feature to be able to add date/time to the "Navigation Bar Header" in settings. 
Time shall contain seconds. Date and time info should be picked up from the server. In such way clock works as a "liveliness" indicator as well. Very common in safety critical systems.

serite, 2022-08-05:
create a script create time info ,and make a tag,  use setTag show the time per sec .

luleipo, 2022-09-08:
Did you solve the problem? Please go public. Thank you

algismo, 2022-09-08:
I was myself planning to extend the client side to allow user to add a tag in the "Header Navigation Bar". Then @serite  proposal could work for showing a time. If somebody will come with any better options, that is very welcomed!

unocelli, 2022-09-08:
@algismo Great,
how about creating it first as a control (e.g. [widget](https://mostafa-kheibary.github.io/Clock.js/))?
let me know if you need some help...

algismo, 2022-10-09:
@unocelli, what do you mean by saying "first as a control"? Can you elaborate more on your thoughts.

I myself envisioned something as simple as this: 
[image]

It could also be an option to add "+" button and allow user to setup multiple such "tags" in the header bar.

unocelli, 2022-10-10:
Hi, yes it would be great to have this feature. 
The top right corner is already reserved for login/logout.
I propose you to create a more flexible interface, in the style of the navigation side, so you can configure label (tag value) or button (function like open card, open page, ...)

algismo, 2022-10-16:
Hi. Ok, thanks for input. Configurable position would be best, agree.
I have to complete another extension that is burning for me as of now, then I will switch back to this.

unocelli, 2022-10-16:
Ok, let me know if you need any help

unocelli, 2023-10-10:
The feature is available in the latest version: Configurable header items: button, label and icons.

closed 2023-10-10 by unocelli

Jue07, 2024-02-22:
> The feature is available in the latest version: Configurable header items: button, label and icons.

Hi can you help how to configure? Thanks in advanced.

unocelli, 2024-03-05:
https://youtu.be/CgxWorE0Fkc?si=MrUlRALQDhPLJ6jJ