frangoteam/FUXA #438: data is not the same in the table

hiwoodpecker, 2022-08-03:
Hi,
when I use table to show data,the data is not the same in another place.i use tag data tag-1 is 9.4,but in the table is 9.
Thanks.
[image]
[image]

algismo, 2022-08-04:
Hi,

what is the expected (correct) value and what do you see in FUXA?

Algirdas

hiwoodpecker, 2022-08-09:
@algismo I want show is 9.4,but it is  9 in the table.
[image]

unocelli, 2022-08-09:
Hi,
how have you set the formatting of the value in the cell? in the table is different.
[image]

hiwoodpecker, 2022-08-11:
@unocelli I known that.Thank you very much.

unocelli, 2022-08-24:
can you show me what you have set?

unocelli, 2022-10-12:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2022-10-12 by unocelli