frangoteam/FUXA #439: ReactJS

sabrine2433, 2022-08-04:
Can fuxa be integrated in a React application as an editor !!??

unocelli, 2022-08-24:
Hi, If you think of the frontend, there is the branch inmation where the frontend is a webcomponent and the communication with the backend is via a bridge. the branch is old but it can be use in a react project.

Bykiev, 2022-10-02:
It's a good idea to provide a separate package with UI-control only. I don't need a NodeJS backend and would like to use only Angular control

Bykiev, 2022-10-13:
@unocelli, what is a branch name?

unocelli, 2022-10-14:
[
[image]
](https://github.com/frangoteam/FUXA/tree/inmation)

Bykiev, 2022-10-14:
> [ [image] ](https://github.com/frangoteam/FUXA/tree/inmation)

Thank you, can you please describe what is done and what need to be done?

unocelli, 2022-10-14:
surely you must have **good angular skills**:
- you have to build the frontend `ng build --configuration=client`
- in dist folder you will find the library of fuxa frontend to use as webcomponent

patsonkaushik, 2022-10-21:
give me link of this branch. i couldnt find it?.

Bykiev, 2022-10-21:
https://github.com/frangoteam/FUXA/tree/inmation

closed 2023-02-05 by unocelli