frangoteam/FUXA #444: Hide content from pipes when stopped

unocelli, 2022-08-07:
### Discussed in https://github.com/frangoteam/FUXA/discussions/443

<div type='discussions-op-text'>

<sup>Originally posted by **mauroalexandre** August  6, 2022</sup>
First of all, thanks for your awesome project. I'm subscribed in patreon for contrib.

I'm making my first SCADA view using Fuxa and when I use pipes, I need hide de content from pipe when stopped to avoid user confusion.

It's possible? How?

Thanks</div>

mauroalexandre, 2022-08-09:
I updated and tested it, it worked perfectly.

Thanks alot.

Rafael-OliveiraBR, 2022-08-23:
Hi, i'm facing the same problem.

OFF:
[image]

ON:
[image]

Pipe settings:
[image]

I'm doing something wrong?
I would like the pipes to be full grey when tag is 0.

mauroalexandre, 2022-08-23:
@Rafael-OliveiraBR update your versio to 1.1.7-750 and use Hide in type.

Rafael-OliveiraBR, 2022-08-23:
> @Rafael-OliveiraBR update your version to 1.1.7-750 and use Hide content in type.

sorry if I'm being noob here, but the last release is the 1.1.17.
How can I keep it updated?
I push some command in npm terminal?

mauroalexandre, 2022-08-23:
> > @Rafael-OliveiraBR update your version to 1.1.7-750 and use Hide content in type.
> 
> sorry if I'm being noob here, but the last release is the 1.1.17. How can I keep it updated? I push some command in npm terminal?

git pull / npm build / npm start

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli