frangoteam/FUXA #449: Button touch not working when Zoom is Enable

mauroalexandre, 2022-08-09:
When Zoom is enabled and you try to use a mobile to open View touch in button not working.

I'm new in the source code, but, maybe we need add (touchend) event in getBindMouseEvent?

Thanks friends.

unocelli, 2022-08-11:
Hi,
not all components are touch compatible :disappointed:
which button do you mean?

mauroalexandre, 2022-08-11:
> Hi, not all components are touch compatible 😞 which button do you mean?

The button from controls (HtmlButton).

mauroalexandre, 2022-09-14:
Fixed in PR https://github.com/frangoteam/FUXA/pull/486/commits/2c8d617c179497292ff7aa8688da71d8c28768c3

GitTurboy, 2023-09-22:
This problem is still exist in v1.1.13-1204. You can repro it by steps below:
1. set zoom to "Manually" in the Layout Settings dialog
2. Create a button on one view which Event click is "Open Dialog" to any other view
3. Browse the view has button by "http://<you address>:1881/view?name=<the view name>" in edge/chrome or your andriod phone. 
4. if you use Edge or Chrome, Press Ctrl+Shift+I open developer tools and then Ctrl+Shift+M turn Device Simulator on.
5. Click the button
6. try to close the dialog opened on step 5. you will find the view loss reponse now. you can not click the "Close" button to close the dialog.