frangoteam/FUXA #450: Rotate SVG shapes based on a tag value

algismo, 2022-08-12:
In some cases there may be a need to rotate SVG shapes/objects based on a tag value to make a more realistic visualization. In my case I wanted to visualize position of an air valve, like show in the image below:

[image]

In commercial SCADA systems this is almost a standard feature, but I could not find any good way how to accomplish this with FUXA. To solve my problem I implemented functionality myself. The result is a new action with some additional parameters. It looks like this:

[image]

If ok, I can make a pull request.

unocelli, 2022-08-13:
Hi, 
of course, it was missing and it would be great to add it

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli