frangoteam/FUXA #451: Input gauge is missing some key functionality

algismo, 2022-08-12:
Continuing on the input gauge topic (after #431). In my opinion it is missing some key functionality. 
To name:
1. Input validation. Now for example operator can write text in a number field. The result is a "facy" value being sent to the PLC. That is quite crazy!
2. No way to format the shown value. For example for the Text control one can define Units and Format digits, but input gauge has nothing like that. I was wondering why. I think both should have the same functionality available. It is tremendously useful.  
3. When "input dialog" is enabled, current value shall be shown side by side with the new value so that operator has full overview.

The result of all those changes could look like this:

[image]
[image]
[image]

unocelli, 2022-08-13:
Great, I like it and it would be nice to add this too.
maybe in the future it will also be useful to add range validation with reference to dynamic values.

unocelli, 2022-10-12:
Is enable in the last release. I am going to close this as resolved.

closed 2022-10-12 by unocelli