frangoteam/FUXA #454: [Sugestion] Item navigation bar

mauroalexandre, 2022-08-17:
Today my project have a lot of elements SVG like engines below 

[image]

It's more easy to me just hide/show SVG to show current state based on TAG values instead of develop one component for each kind of element to change colors.

But, is very hard manipulate the elements in the editor, because the elements have the same position.

I'm tried to find in svgeditor component one way to create one bar for can I select the element based on list, like my draft below

[image]

jcxjcx123, 2022-08-18:
This is a good suggestion, and it's a useful feature, so I made a graphic list.  Users can select elements by clicking on item
[image]

unocelli, 2022-08-19:
I also think it is a great feature.
@jcxjcx123 looks good, what are you waiting for a PR :wink:

jcxjcx123, 2022-08-22:
> I also think it is a great feature. @jcxjcx123 looks good, what are you waiting for a PR 😉

I'd love to share my new features, but it's not done yet, it needs more polish and testing, I'll PR when I'm done😁

haitan981, 2022-10-12:
> > I also think it is a great feature. @jcxjcx123 looks good, what are you waiting for a PR 😉
> 
> I'd love to share my new features, but it's not done yet, it needs more polish and testing, I'll PR when I'm done😁

You done yet ? Is it possible to share?

unocelli, 2022-11-13:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-11-13 by unocelli