frangoteam/FUXA #457: Data archive

AlexandruAndrei3, 2022-08-30:
Hi,
I created an application where I archived 100 tags read from an OPC server and a Sìemens S7-1500 PLC. I found that I have no control over the volume of recorded data because all data is archived without limit and cyclically at one second intervals. I already have a volume of several GB of data for two months, but I would like to be able to archive only the last 7 days. How could I do this?
Thanks.

unocelli, 2022-09-04:
Hi, I added the missing feature for sqlite timeserie value. (Settings->DAQ storage)
let me know if it works

unocelli, 2022-09-08:
going to close this as resolved. let me know if you have any issues.

closed 2022-09-08 by unocelli

AlexandruAndrei3, 2022-09-13:
Hi, sorry for the delay.Thank you for this new feature. I tested cyclic archiving and it works. But I have the following mentions:
1. There is no selectable time range for tags archiving. I notice that only days can be configured.
2. I cannot configure a acquisition time for each individual tag. Now, the tags are archived every second. It would be preferable to be able to configure different acquisition times for each individual tag. Eg: For temperatures I need to set an acquisition time of 3 minutes, for active energy one hour. Acquisition times of 1 s are too fast and unnecessarily load my hdd.