frangoteam/FUXA #462: Proposal - iFrame auto reload intervar

tayfunonbasioglu, 2022-08-30:
Automatic reload time should be set in iframe object.

mauroalexandre, 2022-09-16:
Hello @tayfunonbasioglu .

I created a PR (https://github.com/frangoteam/FUXA/pull/486) to create MODE in scripts, with this feature we can run script in client-side. 

Reading your issue, I improved my source to works with schedule in client side too, so, you can use javascript to reload your iframe in scheduled script with mode CLIENT.

Look the example in my video.

https://user-images.githubusercontent.com/68612845/190549044-5c820fdd-8a25-40c1-9ec0-bc48555b6f22.mp4

I don't know if @unocelli  will accept my PR, but, if yes I'll send another with schedule suport like demonstrated in video.

tayfunonbasioglu, 2022-09-19:
great successful solution.

mauroalexandre, 2022-09-19:
Check #500

serite, 2024-10-07:
in the scripts , use client mode , I use this  script in function , interver set 3 sec. it not works!!!
 but the setInterval 5 sec ,it works!!!  

var iframes = document.getElementsByName('iframe_auto_reload');  //iframe_auto_reload is the iframe name ,not address or tag 
        if (iframes.length > 0) {
            var iframe = iframes[0]; 
            console.log('Iframe found:', iframe);
         
             // it works ,but need call by a click event 
            setInterval(function() {
                iframe.src = iframe.src;
            }, 5000); // 5000 msec = 5sec
        } else {
            console.log('Iframe not found');
        }

unocelli, 2024-10-16:
@serite Hi, I do not confirm the limit issue (3-5 sec.) and in any case the approach is not recommended because you have to manage the timer