frangoteam/FUXA #465: [Question] Multiple customers in same project

mauroalexandre, 2022-09-01:
Is possible I'm use the same server installation for multiple customers, using cookies or something?

I'm found **FUXA/client/src/app/_services/rcgi/resource-storage** and **AppID**, this is the way? how?

Thanks alot.

unocelli, 2022-09-02:
Hi,
No, `rcgi` is to separate the diverse communication interface from frontend to backend :
- `ResDemoService` used in demo version [https://frangoteam.github.io/](https://frangoteam.github.io/) that don't have the FUXA backend
- `ResClientService` used bei project without FUXA backend. The communication is done via a bridge and the FUXA frontend should be build as web component
- `ResWebApiService` is the standard with WebAPI

You should use docker or the user authorization.

mauroalexandre, 2022-09-02:
So, I need one instance for each customer?

For example, I have 3 customers, I need to run 3 diferent docker instances and put in different URI, and use HTTP Access for each URI.

unocelli, 2022-09-02:
You could use the github installation in different folder and change only the web server port...should work.