frangoteam/FUXA #466: [Sugestion] 'add' button on 'Tag Selection'

Rafael-OliveiraBR, 2022-09-02:
Hi,

When selecting tags for components  the 'Tag Selection' window only shows the already existing ones, when creating a project on the fly makes a little hard to always have to do 'Setup > Connections > Edit Device Tags > Add tag' go back to layout, repeat the same process to a new component...

I suggest to add an 'add Tag' icon next to 'cancel button', this can pop up a new windows where you select what physical device you want to create a new tag, or just move us fast to 'Device Settings' page.
[image]

keep up guys, this project rocks! 🤟

unocelli, 2022-10-10:
Hi sorry for the delay.
The feature makes absolute sense, the problem is that there are at least 3 different dialog to configure the tag or topic: 
- Modbus/S7...
- MQTT
- OPCUA

unocelli, 2024-05-22:
is available in the latest version,  let me know if you have any issues.

closed 2024-05-22 by unocelli