frangoteam/FUXA #467: [Question] Is there any way to add a number input control?

rbonilla31, 2022-09-02:
I've been trying to add an input that only allows numbers, but adding a new control is almost impossible for me.

unocelli, 2022-09-02:
Hi,
why not extend the current one by adding the settings `onlyNumber`?

rbonilla31, 2022-09-05:
> 

It could be a way to solve it, but what i really need is a new input with the type="number". I'm adding a screen keyboard and i need to distinguish between a text input and a number input, because if it's a text input it will pop an alphanumeric keyboard, otherwise a numeric keyboard. In conclusion, what i need is a new control or adding some option to the existing input to change the type.

algismo, 2022-09-08:
Hi, you can achieve "number input only" by using changes introduced in #463. You just need to set min and max values to a desired numeric range, and then only numbers will be accepted by the input. That is not maybe the most user friendly way, but it works already.

Otherwise I agree with @unocelli. It is best to extend the same input control if we need more options to limiting the input. 

I was originally in #463 planning to add a drop down where user could choose a type (numeric, string) and then based on the choice either allow to define numeric range or regex for strings. But I dropped this due to time press in my project.

closed 2022-10-12 by unocelli