frangoteam/FUXA #474: TypeError: Cannot read property 'retention' of undefined

balincagabrielalin, 2022-09-08:
I have found an issue :

(node:1520) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'retention' of undefined
    at E:\FUXA\server\runtime\storage\daqstorage.js:107:74
    at new Promise (<anonymous>)
    at Object.checkRetention (E:\FUXA\server\runtime\storage\daqstorage.js:105:12)
    at _clearStorage (E:\FUXA\server\runtime\jobs\cleaner.js:50:35)
    at E:\FUXA\server\runtime\jobs\cleaner.js:24:27
    at new Promise (<anonymous>)
    at Cleaner.execute (E:\FUXA\server\runtime\jobs\cleaner.js:19:16)
    at E:\FUXA\server\runtime\jobs\index.js:168:61
    at Array.forEach (<anonymous>)
    at E:\FUXA\server\runtime\jobs\index.js:166:26
    at new Promise (<anonymous>)
    at _checkJobs (E:\FUXA\server\runtime\jobs\index.js:163:16)
    at _checkStatus (E:\FUXA\server\runtime\jobs\index.js:98:25)
    at Timeout._onTimeout (E:\FUXA\server\runtime\jobs\index.js:31:17)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:1520) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1520) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

unocelli, 2022-09-08:
:weary: Thanks a lot, i have to fix quickly.

haitan981, 2022-10-10:
> 😩 Thanks a lot, i have to fix quickly. The default value is missing. the server still works and if you confirm the setting retention the problem should no longer occur [image]

Did you find a way to fix this error? I want to use SQLite but can't. Only works with InluxDB but low efficiency

unocelli, 2022-10-10:
yes, the problem has been fixed, try it to believe.

unocelli, 2022-10-12:
I am going to close this as resolved. let me know if you have any issues.

closed 2022-10-12 by unocelli