frangoteam/FUXA #477: DAQ to CSV export script/feature

afbenevides, 2022-09-10:
Hello there, I just found your project and it seems really interesting. I searched but could not find a data-acquisition download data kind of feature. Is there already something available for that? I was thinking of a button with parameters to download data for value/tag 1-2-3 from date/time x to date/time y. That would export a CSV file. What do you think? For the rest of the HMI features its already awesome!

Thanks again

unocelli, 2022-09-15:
Hi,
In History table you can export the content in CSV format.

afbenevides, 2022-09-20:
OK thanks for the reply I will test that, that should work for me normally. Thanks again!

afbenevides, 2022-10-18:
@unocelli It really does look to be exactly what we need, but I don't know what I do wrong but no data is showing in the table. Where can we add data to this table in the interface?

Thanks for your help

afbenevides, 2022-10-18:
To make it clearer, the table can display live mata as shown here. See data fits the live graph on same tag. But when in history table (to get the csv download feature), no tags show up... is there a configuration needed other then columns?
I put for reference, table working, history table no data, history table setup window.
[image]
[image]
[image]
[image]

unocelli, 2022-10-18:
Hi, Thanks for the report.
have you configured the tags property to be recorder for the history?
[image]

afbenevides, 2022-10-18:
Hello thanks for the super fast reply. Yes I have put in fact 1 sec interval instead of 60 like in yours. It is the only difference. I can try with 60, should that do a problem?

afbenevides, 2022-10-19:
@unocelli I have an update, I erased one tag of my history table setup, and if I have only time + a tag it works I get history data. As soon as I add another tag it doesn't work anymore.

Is there something you have in mind for me to test to resolve that?

Thanks
[image]
[image]

afbenevides, 2022-10-19:
ok works as intended now. I activated login for the second value also.... Sorry for that...
Thanks this feature was exactly what I was looking for.

closed 2022-10-19 by afbenevides